cornell-zhang/heterocl #380: Problem running hls using the master branch

ezw2, 2021-06-16:
When I run the gemm example, I get this error, though it might just be due to how I'm running it (I just used f = hcl.build(s, target) and f(args)). The project folder gets generated but the report shows all zeroes for latency and resource usage. 

`INFO: [Common 17-206] Exiting vivado_hls at Tue Jun 15 22:54:20 2021...`
`Traceback (most recent call last):`
  `File "gemm.py", line 135, in <module>`
    `time_gemm()`
  `File "gemm.py", line 73, in time_gemm`
    `f(hcl_m1, hcl_m2)`
  `File "/home/ezw2/heterocl/python/heterocl/tvm/_ffi/function.py", line 128, in __call__`
    `return f(*args)`
  `File "/home/ezw2/heterocl/python/heterocl/tvm/_ffi/_ctypes/function.py", line 183, in __call__`
    `ctypes.byref(ret_val), ctypes.byref(ret_tcode)))`
  `File "/home/ezw2/heterocl/python/heterocl/tvm/_ffi/base.py", line 66, in check_call`
    `raise TVMError(py_str(_LIB.TVMGetLastError()))`
`heterocl.tvm._ffi.base.TVMError: TVMCall CFunc Error:`
`Traceback (most recent call last):`
  `File "/home/ezw2/heterocl/python/heterocl/tvm/_ffi/_ctypes/function.py", line 54, in cfun`
    `rv = local_pyfunc(*pyargs)`
  `File "/home/ezw2/heterocl/python/heterocl/tvm/runtime.py", line 110, in tvm_callback_exec_evaluate`
    `out = parse_xml(Project.path, print_flag=True)`
  `File "/home/ezw2/heterocl/python/heterocl/report.py", line 365, in parse_xml`
    `summary = perf_estimate["SummaryOfLoopLatency"]`
`KeyError: 'SummaryOfLoopLatency'`

Hecmay, 2021-06-16:
It seems like a parsing error. The parser was assuming the key should present in the HLS report, which is not always true. do you have any inputs? @yn224

ezw2, 2021-06-16:
I think it's not just a problem with the report parser though, because report.json just had zeros for latency and resource usage

yn224, 2021-07-07:
If I recall correctly, the problem was due to the fact that `.to` primitives were commented out in the code so the code did not know where to perform the simulation. After uncommenting, I get a proper output:
```
+-------------------+-----------------------------------+
| HLS Version       | Vivado HLS 2019.1.3               |
| Product family    | zynq                              |
| Target device     | xc7z020-clg484-1                  |
| Top Model Name    | test                              |
+-------------------+-----------------------------------+
| Target CP         | 10.00 ns                          |
| Estimated CP      | 8.510 ns                          |
| Latency (cycles)  | Min 1056897; Max 1056897          |
| Interval (cycles) | Min 1056898; Max 1056898          |
| Resources         | Type        Used    Total    Util |
|                   | --------  ------  -------  ------ |
|                   | BRAM_18K       0      280      0% |
|                   | DSP48E         3      220      1% |
|                   | FF           203   106400      0% |
|                   | LUT          271    53200      1% |
+-------------------+-----------------------------------+
[00:08:03] Execution complete

+-----------------+--------------+-----------+---------------------+---------------+------------------+
|                 |   Trip Count |   Latency |   Iteration Latency |   Pipeline II |   Pipeline Depth |
|-----------------+--------------+-----------+---------------------+---------------+------------------|
| out_matrix_x    |           64 |   1056896 |               16514 |           N/A |              N/A |
| + out_matrix_y  |           64 |     16512 |                 258 |           N/A |              N/A |
| ++ out_matrix_r |           64 |       256 |                   4 |           N/A |              N/A |
+-----------------+--------------+-----------+---------------------+---------------+------------------+
* Units in clock cycles

```
@Hecmay @ezw2 I think we can close this issue now?

Hecmay, 2021-07-07:
Thanks @yn224. That makes sense. 

@ezw2 can you please run it with .to and see if the error can be solved?

ezw2, 2021-07-07:
Yeah sorry I think it's solved now

closed 2021-07-07 by ezw2