cornell-zhang/heterocl #396: How can I calculate atan2 with HeteroCL?

MariusWiggert, 2021-07-27:
This is probably a really dumb question but unfortunately I couldn't find an answer in the last hours.
Is there an easy way to calculate the atan with heterocl? I found the python API for sin, cos, tanh, but not atan.

Help is highly appreciated! 

Marius

seanlatias, 2021-07-27:
We don't have such API unfortunately. You will need to write your own function to calculate the (approximate) value of arctangent.

seanlatias, 2021-07-27:
One way is to use Taylor expansion. You may find this [article](https://spin.atomicobject.com/2012/04/24/implementing-advanced-math-functions/) useful.

MariusWiggert, 2021-07-28:
Thanks for clarifying @seanlatias ! The article seems actually quite helpful. I found C implementations of arctan but they looked a lot more complicated with look up tables etc.

Is there a way to plug in a C++ or C implementation of arctan into heterocl?

seanlatias, 2021-07-28:
Currently we don't have such a feature. We are also thinking about importing some existing C/C++ implementations into HCL, especially those more hardware-friendly ones.