cornell-zhang/heterocl #406: Segfault when @hcl.def_ is at top-level

jcasas00, 2021-08-11:
Code:

```
hcl.init()

@hcl.def_([(10,), (10,), ()])
def find_max(A, B, x):
    with hcl.if_(A[x] > B[x]):
        hcl.return_(A[x])
    with hcl.else_():
        hcl.return_(B[x])

def maximum(A, B, C, D):
    max_1 = hcl.compute(A.shape, lambda x: find_max(A, B, x), "max_1")
    max_2 = hcl.compute(A.shape, lambda x: find_max(C, D, x), "max_2")
    return hcl.compute(A.shape, lambda x: find_max(max_1, max_2, x), "max_o")

A = hcl.placeholder((10,), "A")
B = hcl.placeholder((10,), "B")
C = hcl.placeholder((10,), "C")
D = hcl.placeholder((10,), "D")

s = hcl.create_schedule([A, B, C, D], maximum)
print(hcl.lower(s))

f = hcl.build(s)
```

Result:
The above code generates a seg fault at the build stage.   If the @hcl.def_  line is commented out, it works fine.   

It is also okay if the whole find_max code with hcl.def_ is moved inside the maximum function (the example in the docs).  This isn't ideal though as it restricts the ability to create modular blocks.

seanlatias, 2021-08-11:
Right. This is a know issue. #387

jcasas00, 2021-08-11:
Yes, same exact problem.  The work-around using a local import should work but is limited to a fresh session (when the local import actually happens).   Can possibly force a reload of the module at the top of the main function so it handles the case where the hcl.def_'ed function is used in multiple places as well.