cornell-zhang/heterocl #408: hcl.print prints tensors with indices in reverse order (as if transposed)

jcasas00, 2021-08-30:
Code:

```
def test (DRAM):
    hcl.print ((), f"DRAM: {DRAM.shape} {DRAM.dtype}\n")
    hcl.print(DRAM)

DRAM    = hcl.placeholder((4,2), "dram", dtype=hcl.UInt(64))
s = hcl.create_schedule([DRAM], test)
f = hcl.build(s)
dram    = hcl.asarray (np.zeros(DRAM.shape), dtype=DRAM.dtype)
print (dram)

```
Output: 
```
[[0 0]                             # output of print(dram) which correctly looks like (4,2)
 [0 0]
 [0 0]
 [0 0]]
DRAM: (4, 2) uint64  
[[0, 0, 0, 0],                    # output of hcl.print which makes it look like (2,4)
[0, 0, 0, 0]]
```

seanlatias, 2021-08-30:
Please check #409

jcasas00, 2021-08-30:
Tested latest master and it looks good now.  Thanks.

closed 2021-08-30 by seanlatias