cornell-zhang/heterocl #413: How to tell if hcl.assert_ fired programmatically?

jcasas00, 2021-09-02:
hcl.assert_ terminates the LLVM execution when it fires.  However, there doesn't seem to be any indication at the top-level that the hcl.assert_ fired.   Can it raise an exception HCLAssert (for example) ?

seanlatias, 2021-09-02:
Hmmmm...  I'll see if I can enable the return value of the top function. If the return value is non-zero then the program is wrong.

seanlatias, 2021-09-02:
@jcasas00 I proposed a solution in #414. Please see if this is similar to what you have in mind.

jcasas00, 2021-09-02:
Yes, this is pretty much what I was envisioning.

seanlatias, 2021-09-04:
Please check #414. Now it is merged.

jcasas00, 2021-09-07:
I tried https://github.com/cornell-zhang/heterocl/pull/414 and it indeed raises an exception.
One thing though, while the assert error message is printed, the string associated with the caught exception seem to be always "[Assert] Assert Failed!!!"  (which we already know by being in the except block).  Is it possible to have the actual assertion message also be passed to the exception object?

For example:
```
    try:
        f(dram, addr)
    except hcl.debug.AssertError as e:
        print ("Got exception: " + str(e))
```

Output:
```
Invalid index=3                                                # this is printed by the hcl.assert_
Got exception: [Assert] Assert Failed!!             # could str(e) also be the same assert_ message string (instead of printing)?
```

seanlatias, 2021-09-09:
Right, ideally we want to return the error message. For now, it would take some extra work to implement that. I'll mark it as enhancement.