cornell-zhang/heterocl #415: Mac OS intallation problem

melodylylin, 2021-09-04:
Hi, I'm currently on MacBook Pro with MacOS BigSur, and I did Conda install wget and cmake, however, I have another error.

`
"vtable for llvm::raw_string_ostream", referenced from:
      TVM::codegen::LLVMModuleNode::GetSource(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in llvm_module.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [lib/libhcl.dylib] Error 1
make: *** [build-tvm] Error 2
`

seanlatias, 2021-09-04:
Are you using your own LLVM or you are using the LLVM installed by HeteroCL?

seanlatias, 2021-09-04:
Can you also try to install from source?

melodylylin, 2021-09-04:
I’m not quite sure, but I think I’m using the one installed by HeteroCL, is there anywhere to check?

melodylylin, 2021-09-05:
I tried to redo the install from source again, and also edit the LLVM config, to let it use the version that is installed by HeteroCL, still have the same error.

seanlatias, 2021-09-05:
Hmmm... Because from the error message it's related to LLVM. Do you have a more complete error message?

seanlatias, 2021-09-05:
Can you also describe how you install from source? You just need to clone the repo and run make. No other things need to be done.

seanlatias, 2021-09-05:
One thing you can also try is to install `g++`. Mac uses clang by default. And when running make, set the `CXX` variable by using `make CXX=/path/to/g++`.

melodylylin, 2021-09-11:
Hi, so I just tried on the MacBook Pro 16 inches with intel chip and same OS system, it works fine. I guess it's the M1 chip problem on MacBook Pro 13 inches.

closed 2021-09-11 by melodylylin