cornell-zhang/heterocl #416: Host code codegen issue

yn224, 2021-09-08:
### Problem Statement
When I tried simulating KNN algorithm for digit recognition example, I noticed that the host code that gets generated produces some errors.

### Error Layout
Modified code (starting from line 405):
```
  hcl.init(dtype_image)
  target = hcl.Platform.aws_f1
  target.config(compiler="vitis", backend="vhls", mode="hw_sim")
  code = top(target)
  hcl_train_images = hcl.asarray(train_images, dtype_image)
  hcl_knn_mat = hcl.asarray(np.zeros((10,3)), dtype_knnmat)
  code(test_images[0], hcl_train_images, hcl_knn_mat)
```
Output:
```
host.cpp: In function ‘int main(int, char**)’:
host.cpp:98:26: error: ‘buffer_test_image’ was not declared in this scope
   err = kernel.setArg(2, buffer_test_image);
                          ^~~~~~~~~~~~~~~~~
host.cpp:98:26: note: suggested alternative: ‘buffer_train_images’
   err = kernel.setArg(2, buffer_test_image);
                          ^~~~~~~~~~~~~~~~~
                          buffer_train_images
host.cpp:99:108: error: no matching function for call to ‘cl::CommandQueue::enqueueMigrateMemObjects(<brace-enclosed initializer list>, int)’
   err = q.enqueueMigrateMemObjects({buffer_knn_mat, buffer_train_images, buffer_test_image}, 0/*from host*/);
                                                                                                            ^
In file included from xcl2.hpp:48:0,
                 from host.cpp:15:
/usr/include/CL/cl2.hpp:7827:12: note: candidate: cl_int cl::CommandQueue::enqueueMigrateMemObjects(cl::vector<cl::Memory>&, cl_mem_migration_flags, cl::vector<cl::Event>*, cl::Event*)
     cl_int enqueueMigrateMemObjects(
            ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/CL/cl2.hpp:7827:12: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘cl::vector<cl::Memory>& {aka const std::vector<cl::Memory, std::allocator<cl::Memory> >&}’
host.cpp:113:120: error: no matching function for call to ‘cl::CommandQueue::enqueueMigrateMemObjects(<brace-enclosed initializer list>, int)’
   err = q.enqueueMigrateMemObjects({buffer_knn_mat, buffer_train_images, buffer_test_image}, CL_MIGRATE_MEM_OBJECT_HOST);
                                                                                                                        ^
In file included from xcl2.hpp:48:0,
                 from host.cpp:15:
/usr/include/CL/cl2.hpp:7827:12: note: candidate: cl_int cl::CommandQueue::enqueueMigrateMemObjects(cl::vector<cl::Memory>&, cl_mem_migration_flags, cl::vector<cl::Event>*, cl::Event*)
     cl_int enqueueMigrateMemObjects(
            ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/CL/cl2.hpp:7827:12: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘cl::vector<cl::Memory>& {aka const std::vector<cl::Memory, std::allocator<cl::Memory> >&}’
```

### Solution
In order to fix this issue, the codegen for the host file should do the following:
- [ ] `buffer_test_image` should be `test_image`.
- [ ] Allocate a buffer for test image so that it can be placed on the device.

Hecmay, 2021-09-20:
@yn224 this should have been solved with the latest commit to the main branch. can you have a try?

yn224, 2021-09-20:
> @yn224 this should have been solved with the latest commit to the main branch. can you have a try?

It seems like the recent commit does not solve the issue, unfortunately. I noticed that the problem was that the buffer for `test_image` is not getting generated. Would you be able to check that the code modification that I addressed above is valid and that this error should not happen?

Hecmay, 2021-09-20:
Sure. I will double-check and let you assp.

Hecmay, 2021-09-21:
@yn224 you are right. the buffer_test_image is missing somehow. I am creating a new PR to fix this issue. 

A quick workaround would be declaring test_image as a tensor of size (1,).

Hecmay, 2021-09-23:
This should have been fixed in https://github.com/cornell-zhang/heterocl/pull/418. Please reopen the issue if the error still exists.

closed 2021-09-23 by Hecmay