cornell-zhang/heterocl #428: Errors on "make -j8"

lloo099, 2021-11-17:
make -C pkgs
make[1]: Entering directory '/home/enai/Desktop/project/heterocl/pkgs'
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[2]: Entering directory '/home/enai/Desktop/project/heterocl/pkgs/cmake'
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/enai/Desktop/project/heterocl/pkgs/cmake'
make[2]: Entering directory '/home/enai/Desktop/project/heterocl/pkgs/llvm'
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
cd /home/enai/Desktop/project/heterocl/build/pkgs/llvm/build; \
/home/enai/Desktop/project/heterocl/build/pkgs/cmake/build/cmake/bin/cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX=/home/enai/Desktop/project/heterocl/build/pkgs/install -DCMAKE_BUILD_TYPE=Release /home/enai/Desktop/project/heterocl/build/pkgs/llvm/src
-- Could NOT find LibXml2 (missing: LIBXML2_LIBRARY LIBXML2_INCLUDE_DIR) 
-- Native target architecture is X86
-- Threads enabled.
-- Doxygen disabled.
-- Go bindings disabled.
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- OCaml bindings disabled.
-- Could NOT find Python module pygments
-- Could NOT find Python module pygments.lexers.c_cpp
-- Could NOT find Python module yaml
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Building with -fPIC
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld
-- Targeting AArch64
-- Targeting AMDGPU
-- Targeting ARM
-- Targeting BPF
-- Targeting Hexagon
-- Targeting Lanai
-- Targeting Mips
-- Targeting MSP430
-- Targeting NVPTX
-- Targeting PowerPC
-- Targeting Sparc
-- Targeting SystemZ
-- Targeting X86
-- Targeting XCore
-- Configuring done
-- Generating done
-- Build files have been written to: /home/enai/Desktop/project/heterocl/build/pkgs/llvm/build
cd /home/enai/Desktop/project/heterocl/build/pkgs/llvm/build; \
make
make[3]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[4]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
[  0%] Built target LLVMBinaryFormat
[  0%] Built target LLVMDemangle
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
[  1%] Built target LLVMTableGen
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
[  1%] Built target llvm_vcsrevision_h
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
[... 660 lines cut ...]
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
[100%] Built target llvm-rtdyld
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
[100%] Built target llvm-lib
[100%] Built target llvm-dlltool
[100%] Built target llvm-ranlib
[100%] Built target sancov
[100%] Built target llvm-xray
[100%] Built target BugpointPasses
[100%] Built target opt
make[4]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
make[3]: Leaving directory '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build'
touch: cannot touch '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build/.built-llvm': Permission denied
Makefile:53: recipe for target '/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build/.built-llvm' failed
make[2]: *** [/home/enai/Desktop/project/heterocl/build/pkgs/llvm/build/.built-llvm] Error 1
make[2]: Leaving directory '/home/enai/Desktop/project/heterocl/pkgs/llvm'
Makefile.pkg.common:6: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/home/enai/Desktop/project/heterocl/pkgs'
Makefile:8: recipe for target 'build-pkgs' failed
make: *** [build-pkgs] Error 2

lloo099, 2021-11-17:
when I run "sudo make -j8", it shows like these:

make -C tvm
make[1]: Entering directory '/home/enai/Desktop/project/heterocl/tvm'
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -fno-rtti -DDMLC_ENABLE_RTTI=0 -DDMLC_USE_FOPEN64=0  -DTVM_LLVM_VERSION= -MM -MT build/llvm/codegen/llvm/codegen_arm.o src/codegen/llvm/codegen_arm.cc >build/llvm/codegen/llvm/codegen_arm.d
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -fno-rtti -DDMLC_ENABLE_RTTI=0 -DDMLC_USE_FOPEN64=0  -DTVM_LLVM_VERSION= -MM -MT build/llvm/codegen/llvm/llvm_common.o src/codegen/llvm/llvm_common.cc >build/llvm/codegen/llvm/llvm_common.d
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -fno-rtti -DDMLC_ENABLE_RTTI=0 -DDMLC_USE_FOPEN64=0  -DTVM_LLVM_VERSION= -MM -MT build/llvm/codegen/llvm/llvm_module.o src/codegen/llvm/llvm_module.cc >build/llvm/codegen/llvm/llvm_module.d
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
g++ -c -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -fno-rtti -DDMLC_ENABLE_RTTI=0 -DDMLC_USE_FOPEN64=0  -DTVM_LLVM_VERSION= -c src/codegen/llvm/codegen_arm.cc -o build/llvm/codegen/llvm/codegen_arm.o
make[1]: execvp: /home/enai/Desktop/project/heterocl: Permission denied
src/codegen/llvm/llvm_common.cc:95:22: error: operator '<' has no left operand
 #if TVM_LLVM_VERSION < 50
                      ^
/bin/sh: 1: /home/enai/Desktop/project/heterocl: Permission denied
Makefile:209: recipe for target 'build/llvm/codegen/llvm/llvm_common.o' failed
make[1]: *** [build/llvm/codegen/llvm/llvm_common.o] Error 1
make[1]: *** Waiting for unfinished jobs....
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -fno-rtti -DDMLC_ENABLE_RTTI=0 -DDMLC_USE_FOPEN64=0  -DTVM_LLVM_VERSION= -MM -MT build/llvm/codegen/llvm/intrin_rule_llvm.o src/codegen/llvm/intrin_rule_llvm.cc >build/llvm/codegen/llvm/intrin_rule_llvm.d
src/codegen/llvm/llvm_module.cc:64:22: error: operator '<=' has no left operand
 #if TVM_LLVM_VERSION <= 60
                      ^~
src/codegen/llvm/llvm_module.cc:71:22: error: operator '<=' has no left operand
 #if TVM_LLVM_VERSION <= 60
                      ^~
src/codegen/llvm/llvm_module.cc:75:24: error: operator '<=' has no left operand
 #elif TVM_LLVM_VERSION <= 90
                        ^~
src/codegen/llvm/llvm_module.cc:86:22: error: operator '<=' has no left operand
 #if TVM_LLVM_VERSION <= 60
                      ^~
[... cut ...]
                 from src/codegen/llvm/intrin_rule_llvm.cc:7:
src/codegen/llvm/./llvm_common.h:10:10: fatal error: llvm/ExecutionEngine/MCJIT.h: No such file or directory
 #include <llvm/ExecutionEngine/MCJIT.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:209: recipe for target 'build/llvm/codegen/llvm/intrin_rule_llvm.o' failed
make[1]: *** [build/llvm/codegen/llvm/intrin_rule_llvm.o] Error 1
make[1]: *** wait: No child processes.  Stop.
Makefile:11: recipe for target 'build-tvm' failed
make: *** [build-tvm] Error 2

Hecmay, 2021-11-18:
HI @lloo099. I believe the error is caused by the permission issues. You should make sure you have the right permission for those folders. And you do not have to use sudo for the installations.

lloo099, 2021-11-19:
> HI @lloo099. I believe the error is caused by the permission issues. You should make sure you have the right permission for those folders. And you do not have to use sudo for the installations.

Okay, great. ur right
Here are my codes:
```

sudo make clean
chmod 764  ./heterocl
cd ./heterocl
make -j8
```

lloo099, 2021-11-19:
Sorry, it still has problems with permission. But I already use command"chmod 777 .." Could you help me with it? @seanlatias @Hecmay 

```
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -MM -MT build/codegen/hlsc/codegen_vhls.o src/codegen/hlsc/codegen_vhls.cc >build/codegen/hlsc/codegen_vhls.d
g++ -c -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -c src/codegen/hlsc/codegen_vhls.cc -o build/codegen/hlsc/codegen_vhls.o
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -MM -MT build/codegen/hlsc/vhls_module.o src/codegen/hlsc/vhls_module.cc >build/codegen/hlsc/vhls_module.d
g++ -c -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -c src/codegen/hlsc/vhls_module.cc -o build/codegen/hlsc/vhls_module.o
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -MM -MT build/codegen/hlsc/codegen_hlsc.o src/codegen/hlsc/codegen_hlsc.cc >build/codegen/hlsc/codegen_hlsc.d
g++ -c -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -c src/codegen/hlsc/codegen_hlsc.cc -o build/codegen/hlsc/codegen_hlsc.o
src/codegen/hlsc/vhls_module.cc: In lambda function:
src/codegen/hlsc/vhls_module.cc:363:13: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
       system("g++ main.cpp -o out");
       ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
src/codegen/hlsc/vhls_module.cc:365:13: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
       system("./out");
       ~~~~~~^~~~~~~~~
src/codegen/hlsc/vhls_module.cc:367:13: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
       system("rm out main.cpp");
       ~~~~~~^~~~~~~~~~~~~~~~~~~
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -MM -MT build/codegen/hlsc/codegen_ihls.o src/codegen/hlsc/codegen_ihls.cc >build/codegen/hlsc/codegen_ihls.d
g++ -c -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -c src/codegen/hlsc/codegen_ihls.cc -o build/codegen/hlsc/codegen_ihls.o
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -MM -MT build/codegen/spirv/intrin_rule_spirv.o src/codegen/spirv/intrin_rule_spirv.cc >build/codegen/spirv/intrin_rule_spirv.d
g++ -c -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -c src/codegen/spirv/intrin_rule_spirv.cc -o build/codegen/spirv/intrin_rule_spirv.o
g++ -std=c++11 -Wall -O2 -Iinclude -I/home/enai/Desktop/project/heterocl/tvm/dlpack/include -I/home/enai/Desktop/project/heterocl/tvm/dmlc-core/include -IHalideIR/src -fPIC -DTVM_CUDA_RUNTIME=0 -DTVM_ROCM_RUNTIME=0 -DTVM_OPENCL_RUNTIME=0 -DHCL_SDACCEL_RUNTIME=1 -DHCL_VHLS_RUNTIME=1 -DTVM_VULKAN_RUNTIME=0 -DTVM_OPENGL_RUNTIME=0 -MM -MT build/codegen/spirv/codegen_spirv.o src/codegen/spi [... cut ...]
src/codegen/llvm/./././llvm_common.h:10:10: fatal error: llvm/ExecutionEngine/MCJIT.h: No such file or directory
 #include <llvm/ExecutionEngine/MCJIT.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:209: recipe for target 'build/llvm/codegen/llvm/codegen_arm.o' failed
make[1]: *** [build/llvm/codegen/llvm/codegen_arm.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/enai/Desktop/project/heterocl/tvm'
Makefile:11: recipe for target 'build-tvm' failed
make: *** [build-tvm] Error 2

```

Hecmay, 2021-11-19:
@lloo099, Just want to know how you changed the permission to that folder? Can you please try to use the fixes mentioned in this link? https://lanvu.wordpress.com/2011/08/05/fix-the-error-make-execvp-permission-denied/

If that error still occurs, can you please try running `ls -l` to show the permission and owners of your folders.

lloo099, 2021-11-19:
> @lloo099, Just want to know how you changed the permission to that folder? Can you please try to use the fixes mentioned in this link? [...]
Yes, same error:
`drwxr-xr-x 15 enai enai     4096 Nov 19 23:39 heterocl`

Hecmay, 2021-11-21:
@lloo099 I am not fully sure why you still have the permission issues, even after you switch to sudo. But as a workaround, I would suggest you to install HeteroCL in docker container.