cornell-zhang/heterocl #482: git submodule fails

oroppas, 2023-02-24:
I followed the instruction in `README.md` and got

```console
ryuta@orin:~/packages/heterocl/heterocl-mlir/heterocl$ git submodule update --init --recursive
Cloning into '/home/ryuta/packages/heterocl/heterocl-mlir/heterocl/hcl-dialect'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:cornell-zhang/hcl-dialect.git' into submodule path '/home/ryuta/packages/heterocl/heterocl-mlir/heterocl/hcl-dialect' failed
Failed to clone 'hcl-dialect'. Retry scheduled
```

Am I missing something?

chhzh123, 2023-02-24:
Hi, please follow [this guide](https://docs.github.com/en/authentication/troubleshooting-ssh/error-permission-denied-publickey) to make sure you have correctly uploaded your ssh key to Github and retry. Otherwise, you can try using the `http` protocol to clone the submodule.

oroppas, 2023-02-26:
Thank you for the tip! I was able to clone the submodules.

closed 2023-02-26 by oroppas