cornell-zhang/heterocl #507: submodule seems unable to access

quantrpeter, 2023-12-30:
```
/home/peter/2TB/workspace/heterocl-mlir>git submodule update --init --recursive
Cloning into '/media/2TB/workspace/heterocl-mlir/hcl-dialect'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:cornell-zhang/hcl-dialect.git' into submodule path '/media/2TB/workspace/heterocl-mlir/hcl-dialect' failed
Failed to clone 'hcl-dialect'. Retry scheduled
Cloning into '/media/2TB/workspace/heterocl-mlir/hcl-dialect'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:cornell-zhang/hcl-dialect.git' into submodule path '/media/2TB/workspace/heterocl-mlir/hcl-dialect' failed
Failed to clone 'hcl-dialect' a second time, aborting
```

chhzh123, 2023-12-30:
Hi, it seems you need to add the public SSH key to your Github account. You can refer to this [article](https://docs.github.com/en/authentication/troubleshooting-ssh/error-permission-denied-publickey) to set up.

quantrpeter, 2023-12-30:
thanks, these works:

Open Git Bash.
Generate Public Key:

ssh-keygen -t rsa -b 4096 -C "youremailaddress@xyz.com"
Copy generated key to the clipboard (works like CTRL+C)

clip < ~/.ssh/id_rsa.pub
Browser, go to Github => Profile=> Settings => SSH and GPG keys => Add Key
Provide the key name and paste clipboard (CTRL+V).
Finally, test your connection (Git bash)

ssh -T git@github.com

closed 2023-12-30 by quantrpeter