cornell-zhang/heterocl #512: Failure on Running the Python test

anniezfy, 2024-02-19:
I have encountered a problem on the last installation steps.
none of the test cases passed successfully~
The ERROR message shows the problem lies in the missing module name `hcl_mlir`.
I would  greatly appreciate it if someone could help me fix the problem
```
_________________________________ ERROR collecting tests/mlir/test_memory.py _________________________________
ImportError while importing test module '/home/anniezfy/文档/GitHub/heterocl/tests/mlir/test_memory.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.10/importlib/__init__.py:126: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/mlir/test_memory.py:5: in <module>
    import heterocl as hcl
heterocl/__init__.py:5: in <module>
    from .schedule import Schedule, customize, create_schedule
heterocl/schedule.py:7: in <module>
    from hcl_mlir.exceptions import (
E   ModuleNotFoundError: No module named 'hcl_mlir'
```

zzzDavid, 2024-02-19:
Hi, according to the error message, the hcl-dialect project either did not build successfully or wasn't installed successfully. Could you perform the following debugging steps and determine which step did not complete successfully:

1. Look for error message during the building process: is there any error like "LLVM make failed" or "HCL-MLIR make failed", or "HCL-MLIR Python package installation failed"?
2. Check if LLVM was built successfully: check if `heterocl/hcl-dialect/externals/llvm-project/build/bin/mlir-opt` exists. 
3. Check if the following path exists: `heterocl/hcl-dialect/build/tools/hcl/python_packages/hcl_core`. If it doesn't exist, that means hcl-dialect Python module wasn't built successfully.

anniezfy, 2024-02-19:
Thank you so much for your reply~ I performed your debugging suggestions looking for error messages in the build process. HCL-Dialect was successfully built but wasn't installed properly because no write file permission. Creating a virtual python environment for the build process did work!

zzzDavid, 2024-02-19:
Awesome! Glad to know that it's resolved.

yuyuli1, 2025-02-24:
Hi, I don't have the files for either of the two checking methods you mentioned in D:\pycode\DUZIYU\DUZIYU\heterocl-mlir\hcl-dialect\externals\llvm-project\build, so what should I do?
@zzzDavid

chhzh123, 2025-02-24:
Hi, thanks for trying HeteroCL. The project is no longer actively maintained and has been superseded by Allo (https://github.com/cornell-zhang/allo). We recommend using the new framework instead.